ARMmbed/mbed-os-tools #83: Unable execute any test case using greentea command

UrmilaKL, 2019-01-11:
If i try to execute any testcase using below command it is failing with below error, I have  created dummy main.cpp in TESTS\Group1\TestCase1\ and tried to execute it
C:\mbedOSExample\mbed-os-example-blinky>mbedgt -n tests* -t GCC_ARM -m DISCO_L475VG_IOT01A
mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: using multiple test specifications from current directory!
        using 'BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json'
mbedgt: detecting connected mbed-enabled devices...
[1547031484.20][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
mbedgt: detected 1 device
mbedgt: shuffle seed: 0.1498640650
mbedgt: no platform/target matching tests were found!
mbedgt: no matching platforms were found!
mbedgt: completed in 0.96 sec
mbedgt: exited with code -110

C:\mbedOSExample\mbed-os-example-blinky>mbedhtrun --version
1.4.4

C:\mbedOSExample\mbed-os-example-blinky>mbedgt -l
mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: using multiple test specifications from current directory!
        using 'BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json'
mbedgt: available tests for built 'DISCO_L475VG_IOT01A-GCC_ARM', location 'BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM'

bridadan, 2019-01-16:
Hi @UrmilaKL,

These are the steps I use to test with Mbed OS and greentea:

```
git clone https://github.com/ARMmbed/mbed-os
cd mbed-os
mbed test --target DISCO_L475VG_IOT01A --toolchain GCC_ARM
```

The `mbed test` command above should compile and take care of the flashing and communication with the board.

UrmilaKL, 2019-01-16:
@bridadan ,

Thank you for the updates Bridadan, Can i use blinky application to run any test case using green-tea command? ..

bridadan, 2019-01-16:
> Can i use blinky application to run any test case using green-tea command

No you cannot. The blinky application does not include the test library.

You currently can't run tests when your application also has a `main()` function. You may either clone mbed-os again (as above) and run the tests from there OR you can comment out/remove the file that has the `main` function.

UrmilaKL, 2019-01-17:
Hi @bridadan ,

By cloning it i am able to run the test cases now, In my test case i have added few print statements i am unable to see that in the console logs,

UrmilaKL, 2019-01-17:
Below is my test case,
#include "mbed.h"
#include "unity.h"
#include "utest.h"
#include "rtos.h"
#include <events/mbed_events.h>
#include <mbed.h>
#include "greentea-client/test_env.h"
#include "ble/BLE.h"

using namespace utest::v1;

// A test that returns successfully is considered successful
void Display()
{
	printf("Welocome to my home");
	//BLE &ble = BLE::Instance();
	
}
utest::v1::status_t test_setup(const size_t number_of_cases) {
    // Setup Greentea using a reasonable timeout in seconds
    GREENTEA_SETUP(40, "default_auto");
    return verbose_test_setup_handler(number_of_cases);
}
void start_advertising() {
        
	printf("I am calling start advertsing");
}

Case cases[] = {
	
    Case("Display test", Display),
    //Case("Testing failure test", test_failure),
};

Specification specification(test_setup, cases);

int main()
{
	printf("I am printing main");
	BLE &ble = BLE::Instance();
	ble.startAdvertising;
	return !Harness::run(specification);
}
What may be wrong in the above test case?

bridadan, 2019-01-17:
If you're using the `mbed test` command above, just add the `-v` (verbose) argument:

```
mbed test --target DISCO_L475VG_IOT01A --toolchain GCC_ARM -v
```

bridadan, 2019-01-18:
Hi @UrmilaKL, did using the verbose argument (`-v`) fix your issue?

UrmilaKL, 2019-01-21:
@bridadan ,

I am able to see the print statements using -v option, 
Thanks Much

Regards
Urmila Bhat