ARMmbed/mbed-os-tools #84: Issue for the colorama's version

heejungPark, 2019-01-11:
We are trying to test pelion-enablement.
But the command, `mbed test -t ARM -m SDT64B -n simple-mbed-cloud-client-tests-* --run -v`, is typed, the error is occurred like below.
Our colorama's version is 0.4.1. So we think it caused this error.
Is the only solution for this error lower colorama's version than 0.4.0??

```
C:\mbed\Workspace\Test\pelion-enablement>mbed test -t ARM -m SDT64B -n simple-mbed-cloud-client-tests-* --run -v
[mbed] Working path "C:\mbed\Workspace\Test\pelion-enablement" (program)
[mbed] Exec "mbedgt --test-spec .\BUILD\tests\SDT64B\ARM\test_spec.json -n simple-mbed-cloud-client-tests-* -V" in "C:\mbed\Workspace\Test\pelion-enablement"
Traceback (most recent call last):
  File "C:\mbed\Tools\mbed-CLI\Python27\Scripts\mbedgt-script.py", line 6, in <module>
    from pkg_resources import load_entry_point
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 3088, in <module>
    @_call_aside
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 3072, in _call_aside
    f(*args, **kwargs)
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 3101, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 576, in _build_master
    return cls._build_from_requirements(__requires__)
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 589, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "c:\mbed\tools\mbed-cli\python27\lib\site-packages\pkg_resources\__init__.py", line 778, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'colorama<0.4,>=0.3' distribution was not found and is required by mbed-greentea
[mbed] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "C:\mbed\Workspace\Test\pelion-enablement"
       Command: "mbedgt --test-spec .\BUILD\tests\SDT64B\ARM\test_spec.json -n simple-mbed-cloud-client-tests-* -V"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

bridadan, 2019-01-15:
Hi @heejungPark, any update?

heejungPark, 2019-01-16:
Thanks to you, there are no additional error only for package tool's version.
But for processing on mbed test, there is another issue.
I don't know why it occurs...
Could you please check attached log file?
[Log.txt](https://github.com/ARMmbed/mbed-os-tools/files/2763569/Log.txt)

bridadan, 2019-01-16:
Hi @heejungPark, I'm afraid I don't know why your particular target is working. You can see the general process for how the test tools will flash and reset the target here: https://github.com/ARMmbed/mbed-os-tools/tree/master/packages/mbed-host-tests#htrun-mbed-host-tests

Please verify that those methods work for your device. I will close this issue since the colorama issue has been resolved, please open a new issue if you believe there is an issue with the testing process.

closed 2019-01-16 by bridadan