ARMmbed/mbed-os-tools #94: List ST boards under "stlink" section in DEFAULT_PLATFORM_DB

RomanSaveljev, 2019-01-11:
I was wondering why ST-boards are listed under "daplink" section in the `DEFAULT_PLATFORM_DB`? The `device_type` seems to be correctly resolving to "stlink" from VID. Turns out mbed-flasher depends on the right content in `DEFAULT_PLATFORM_DB`, when choosing the right flashing method.

I am testing against **v1.6.2**, the behavior in e.g. **v1.4.6** is different (ST boards tell they are "daplink").

RomanSaveljev, 2019-01-17:
A small update. The changes on our side that depend on this patch have been relocated farther in the backlog. We will do something else before coming back to testing locally patched mbedls.

RomanSaveljev, 2019-01-24:
We have the patch under testing (https://github.com/RomanSaveljev/mbed-ls/pull/2). Looks promising.

bridadan, 2019-02-01:
@RomanSaveljev do you still want to use the patch you've posted above?

RomanSaveljev, 2019-02-03:
We no longer seem to depend on it.

bridadan, 2019-02-04:
OK, closing for now then.

closed 2019-02-04 by bridadan