ARMmbed/mbed-os-tools #98: .mbedignore files

Alex-EEE, 2019-01-16:
We're just now upgrading to mbedos 11...we used to keep one .mbedignore file in the root directory, where we would skip some tests that weren't passing on our system.  Since the upgrade, it seems greentea only skips tests if the .mbedignore file is under the TESTS directory.  does that sound right?  Where will greentea look for .mbedignore files in the directory tree?

bridadan, 2019-01-17:
@Alex-EEE that doesn't sound quite right. Unless of course we introduced a bug 😄 

Greentea sees any tests that were built. If the tests were built (and added to the test specification file in your BUILD directory), then it will be ran unless you filter it with `-n` option.

Could you post the contents of your `.mbedignore` file and the general structure of your project? I can try to reproduce the issue.

Alex-EEE, 2019-01-17:
Had to remove the leading dot and add txt to get it to upload.  This sits in the mbed folder, the parent of mbed-os.  The only other weird thing we do is supply --path and --source as ".." so that we also scan the parent folder of mbed for TESTS.  Here is what our commands look like:
```
mbed test --paths .. --source .. --compile --clean --app-config ../test-server-dependencies/k64/mbed_app.json
mbed test --run --copy jlink  -r jlink --target=K64F -v
```

Thanks!  

[mbedignore.txt](https://github.com/ARMmbed/mbed-os-tools/files/2769750/mbedignore.txt)

bridadan, 2019-01-17:
So your project folder looks like this?

```
/project
    /TESTS
    /mbed
        /.mbedignore
        /mbed-os
```

And you're executing Mbed CLI from the `mbed` folder?

Alex-EEE, 2019-01-17:
Close.  ```
\code
  \ feature1
     \TESTS
\mbed
  .mbedignore
  \mbed-os 
```

Yea, executing from mbed folder

bridadan, 2019-01-17:
Ok thanks. I reproduced the scenario, it does look like the .mbedignore file does have to be inside the TESTS directory in order for the tests to be skipped. I tried this in Mbed OS 5.10.4 and 5.11.1.

So while the behavior doesn't seemed to have changed, I still think this is a bug. This is an issue with the Mbed OS build system though, so the issue would need to go on the main Mbed OS repo.

Alex-EEE, 2019-01-17:
Thanks for looking into it!  Can you move the issue?  ( You'll probably get more attention than me haha )

bridadan, 2019-01-17:
I can't actually "move" issues to `mbed-os`, I'd just be copying and pasting stuff. You'll get plenty of attention don't worry, you can tag me :)

bridadan, 2019-01-18:
Thanks for raising this in mbed-os! https://github.com/ARMmbed/mbed-os/issues/9419

Closing here.

closed 2019-01-18 by bridadan