ARMmbed/mbed-os-tools #111: Unable to communicate with BLE APIs through greentea testing

UrmilaKL, 2019-01-23:
Description: Unable to communicate with BLE APIs through greentea testing 

Target: DISCO_L475VG_IOT01A
Toolchain: GCC_ARM
mbed-cli version: 1.8.3
mbed-os->5.11.1

Steps to Reproduce:
1. Create directory structure like below
inside mbed-os -> TESTS->Group1->TestCase1->main.cpp(I have attached my main.cpp which i tried)
2. Create the BLE object like below
BLE &ble = BLE::Instance();
ble.gap().startAdvertising();
3. Add all necessary (BLE.h, BLE.cpp... ) files to the same directory where you are running the test case
2. Try to run the testcase using below command
mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM -n tests-Group1-TestCase1

Observation: Other existing test cases are passing, When we try to create BLE object and try to call any API it is failing with the below error

"Please provide the implementation for MBED BLE"

Please find the attached logs and main.cpp file

bridadan, 2019-01-23:
@UrmilaKL I believe you need to enable the `BLE` feature. You can do this by adding an `mbed_app.json` file in your project root and adding the contents shown here: https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_Privacy/mbed_app.json#L11-L14

(You only need the entry for the DISCO_L475VG_IOT01A)

bridadan, 2019-01-25:
Hi @UrmilaKL, any update?

UrmilaKL, 2019-01-25:
Hi @bridadan ,

closed 2019-01-25 by UrmilaKL

UrmilaKL, 2019-02-05:
Hi @bridadan ,

The above code for BLE is already added in the mbed_app.json file

bridadan, 2019-02-05:
> Do we have any specific test cases for green tea and ice tea for BLE examples in ARM mbed ?

I personally don't work on those tests so I don't know. I would ask on the https://github.com/armmbed/mbed-os repository.

UrmilaKL, 2019-02-06:
Sure @bridadan

UrmilaKL, 2019-02-07:
Hi @bridadan ,
Did you get any response for this issue, I am trying to call BLE API for any BLE app from mbed community and failed with below issue
"Please provide the implementation for MBED BLE"
Thanks
Urmila Bhat

bridadan, 2019-02-07:
Hi @UrmilaKL, sorry I wasn't clear before, I was asking **you** to file the issue directly on the https://github.com/armmbed/mbed-os repository. That was you can speak directly with the people who own that part of the project. I don't think I'll be a helpful middle man in this case 😄