ARMmbed/mbed-os-tools #117: [DISCO_L475VG_IOT01A]LED Blinker: Fails to connect with LED app

UrmilaKL, 2019-01-25:
Description: [DISCO_L475VG_IOT01A]LED Blinker: Fails to connect with LED app

Target: DISCO_L475VG_IOT01A
Toolchain: GCC_ARM
mbed-cli version: 1.8.3
mbed-os->5.11.1

App used: mbed-os-example-ble-LEDBlinker

Steps to Reproduce:
Import project from https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-ble-LEDBlinker/
Edit the mbed-os.lib file to below link to
https://github.com/ARMmbed/mbed-os/#2fd0c5cfbd83fce62da6308f9d64c0ab64e1f0d6
Do mbed deploy to update mbed-os to use the above revision
Do mbed compile -target DISCO_L475VG_IOT01A -toolchain GCC_ARM
In another setup make LED app ready and should be advertising 

Expectation: Soon after download LEDBlinker app should connect with LED app

Observation: LEDBlinker connection with LED is not successful, Also in the teraterm unable to see any logs (apart from MAC device BD Address)

Note: Recently they have made some changes to the code, LEDBlinker older version of app is working fine, We have downloaded the latest version of  app from https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-ble-LEDBlinker/ site

jeromecoutant, 2019-01-29:
Hi

Any update ?

Thx

jeromecoutant, 2019-01-29:
@avilei @apalmieriGH

It seems that BLE is not supporting the deep sleep mode ?
Do you have any advice where we should disable it ?
(ex: https://github.com/ARMmbed/mbed-os/pull/8219)

linlingao, 2019-02-22:
@UrmilaKL It seems you got LED working with mbed_app.json update. Is this issue solved?

closed 2019-02-28 by ciarmcom