ARMmbed/mbed-os-tools #119: [DISCO_L475VG_IOT01A]GAP Button: Service data field is not changing upon pushing USER button

UrmilaKL, 2019-01-29:
Description: [DISCO_L475VG_IOT01A]GAP Button: Service data field is not changing upon pushing USER button
Target: DISCO_L475VG_IOT01A
Toolchain: GCC_ARM
mbed-cli version: 1.8.3
mbed-os->5.11.1
App used: mbed-os-example-ble-GAPButton

Steps to Reproduce:
Import project from https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-ble-GAPButton/
Edit the mbed-os.lib file to below link to
https://github.com/ARMmbed/mbed-os/#2fd0c5cfbd83fce62da6308f9d64c0ab64e1f0d6
Do mbed deploy to update mbed-os to use the above revision
Do mbed compile -target DISCO_L475VG_IOT01A -toolchain GCC_ARM
Scan the device from nRF or lightblue  peer app
Push on the APP button from the device to see whether it reflects the button press count from starting value 0 in the Service Data field

Expectation: The Service Data field value of the advertisement packet should change every time you press the button.
Observation: Service data value is not changing when we press the APP button from the device

bridadan, 2019-01-29:
Hi @UrmilaKL,

Thanks for raising the issue, though may I request that in the future if you have questions about examples you open these issues against the example repositories? For the BLE examples this place is here: https://github.com/ARMmbed/mbed-os-example-ble

Mbed OS applications pin the OS version because this is the version they are tested again. If you change the version of the OS, the application may or may not work correctly (due to changes in APIs). In this case, there is a revision that uses 5.10.4 that you should use: https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-ble-GAPButton/rev/49aa083ab86d/ (you can see the revision at the bottom of the page). Please be sure to change your whole application to this version if you need to move Mbed OS back to 5.10.4.

desmond-blue, 2019-02-11:
Hi @UrmilaKL ,

There is already an issue opened in project [mbed-os-example-ble](https://github.com/ARMmbed/mbed-os-example-ble/issues/215), please follow this issue there.

bridadan, 2019-02-11:
Thanks @desmond-blue, closing this one.

closed 2019-02-11 by bridadan