ARMmbed/mbed-os-tools #120: Getting Traceback error while performing mbedls with mbedls-1.7.1 version in UBUNTU 16.04 LTS amd MAC OS(10.13.6)

Vanireddy6, 2019-01-30:
**Steps to reproduce**:
1.Install mbedls - 1.7.1 version.
2.Do mbedls and observe below error.
 mbedls --version
Traceback (most recent call last):
  File "/home/cy-user/.local/bin/mbedls", line 10, in <module>
    sys.exit(mbedls_main())
  File "/home/cy-user/.local/lib/python2.7/site-packages/mbed_lstools/main.py", line 183, in mbedls_main
    logger.debug("mbed-ls ver. %s", get_version())
  File "/home/cy-user/.local/lib/python2.7/site-packages/mbed_lstools/main.py", line 41, in get_version
    return pkg_resources.require("mbed-ls")[0].version
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 943, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 834, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (urllib3 1.19.1 (/usr/local/lib/python2.7/dist-packages/urllib3-1.19.1-py2.7.egg), Requirement.parse('urllib3<1.25,>=1.21.1'), set(['requests'])).

bridadan, 2019-01-30:
Hi @Vanireddy6,

Looks like you have some conflicting packages on your system:
```
pkg_resources.ContextualVersionConflict: (urllib3 1.19.1 (/usr/local/lib/python2.7/dist-packages/urllib3-1.19.1-py2.7.egg), Requirement.parse('urllib3<1.25,>=1.21.1'), set(['requests'])).
```

Have you tried updating your version of urllib3? `pip install -U urllib3`.

Vanireddy6, 2019-01-31:
Hi @bridadan ,

Yes. Already updated urllib3 version and tried. Still seeing the same issue.

bridadan, 2019-01-31:
I can't reproduce this issue on my machine. What I ran (on Windows 10 with Python 2.7):

```
pip uninstall mbed-ls
pip install mbed-ls
mbedls --version
```

Would you mind posting the output from `pip list`?