ARMmbed/mbed-os-tools #128: Question about the stability of this repo

sathysreenath, 2019-02-13:
### Description

### Issue request type

    [ x] Question
    [ ] Enhancement
    [ ] Bug

bridadan, 2019-02-13:
Hi @sathysreenath, looks like your question may have been dropped. In general if your question is "what can I consider stable" you can refer to the chart in our readme: https://github.com/ARMmbed/mbed-os-tools#packages

The python APIs/command line of `mbed-ls`, `mbed-host-tests`, and `mbed-greentea` are stable, so you can depend on those. But the API of `mbed-os-tools` is currently unstable.

bridadan, 2019-03-05:
@sathysreenath I will close this as I believe I've answered your question, please let me know if I can clarify this further.

closed 2019-03-05 by bridadan