ARMmbed/mbed-os-tools #137: [OS ERROR] Command: armclang --vsn : Unable to build (with latest mbed-os) application and tests with ARM compiler 5

fahim-ublox, 2019-03-11:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [*] Bug

fahim-ublox, 2019-03-11:
@screamerbg @RobMeades

wajahat-ublox, 2019-03-11:
I think the issue is coming after [this](https://github.com/ARMmbed/mbed-os/commit/50eb4f9da88cfe6bad4e05571168199da5b01638#diff-8ae3dde46b55f3a922724cd3585922b2) commit.

I may be wrong but looks like there is a mistake in the following line, the IF statement should be `and` instead of `or`. The method always returns "ARMC6" if "ARM" is in target.supported_toolchains.
[`if("ARM" in target.supported_toolchains or "ARMC6" in target.supported_toolchains)`](https://github.com/ARMmbed/mbed-os/blob/ba24cb28752b17b597d152e6abf13cc19e3fc6d6/tools/build_api.py#L126)

On a side note, if i define ARMC5 instead of ARM in targets.json for C030, then it builds fine
`"supported_toolchains": ["GCC_ARM", "ARMC5", "IAR"],`

bridadan, 2019-03-11:
Hi @fahim-ublox and @wajahat-ublox. As part of the upcoming 5.12.0 release, Mbed OS is switching to use Arm Compiler 6: https://github.com/ARMmbed/mbed-os/pull/9888. The change you've posted is intentional. You can force the tools to use ARMC5 by doing the change you posted above, however it is recommended to use Arm Compiler 6 going forward.

wajahat-ublox, 2019-03-12:
Hi @bridadan. I understand that ARMC6 is now default option but just for the sake of clarification, if a target supports both ARMC5 and ARMC6, how can user tell mbed-cli which compiler to use? There is a chance that user has a license for ARMC5 but not ARMC6.

Secondly, mbed-cli lists the following toolchains as supported so it gets a little confusing whether to give -t ARM or -t ARMC6
`ARM,     ARMC6,   GCC_ARM, IAR,     uARM`

bridadan, 2019-03-12:
> Secondly, mbed-cli lists the following toolchains as supported so it gets a little confusing whether to give -t ARM or -t ARMC6
> `ARM, ARMC6, GCC_ARM, IAR, uARM`

The intention is to phase out ARMC5 support I believe and to emphasize ARMC6. So supporting both compilers with out tools and Mbed CLI is a temporary measure until all the targets are brought up to speed.

> I understand that ARMC6 is now default option but just for the sake of clarification, if a target supports both ARMC5 and ARMC6, how can user tell mbed-cli which compiler to use? There is a chance that user has a license for ARMC5 but not ARMC6.

It's pretty much how you did it above. You need to edit the `targets.json` or override it in `mbed_app.json`.

I would however like to re-state that as far as Mbed OS goes, targets won't officially support ARMC5 and ARMC6 at the same time since we aren't currently planning on testing both of them. So the ARMC5 build could break at any time in the future.

wajahat-ublox, 2019-03-13:
Understood. Thanks for the clarification.

screamerbg, 2019-03-13:
> Hi @fahim-ublox and @wajahat-ublox. As part of the upcoming 5.12.0 release, Mbed OS is switching to use Arm Compiler 6: [ARMmbed/mbed-os#9888](https://github.com/ARMmbed/mbed-os/pull/9888). The change you've posted is intentional. You can force the tools to use ARMC5 by doing the change you posted above, however it is recommended to use Arm Compiler 6 going forward.

@bridadan I just checked with the Mbed product management and according to them, we are not "switching" to ARM Compiler 6 - we are making "Arm Compiler 6" the default ARM Compiler. We are also keeping support for ARM Compiler 5 as one of the supported compilers until  the support for it is officially announced as deprecated by the **Product Management** (not random Mbed  engineer). In fact, we have a number of Partners and Customers that will continue to use ARM Compiler 5 with Mbed OS 5.12 as a main vehicle for their Products (check with @ChiefBureaucraticOfficer), including a customer of U-blox.

closed 2019-03-28 by screamerbg