ARMmbed/mbed-os-tools #141: parse_global_resource_mgr(...) default port = None -> ':'.join(grm_values[1:]) -> TypeError

alekla01, 2019-03-19:
### Description

port_name = None, https://github.com/ARMmbed/mbed-os-tools/blob/master/src/mbed_os_tools/test/mbed_test_api.py#L582
':'.join(grm_values[1:]), https://github.com/ARMmbed/mbed-os-tools/blob/master/packages/mbed-greentea/mbed_greentea/mbed_greentea_cli.py#L697
-> TypeError: sequence item 2: expected string, NoneType found

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

bridadan, 2019-03-19:
Hi @alekla01, could you please provide the command you ran to get this error?

alekla01, 2019-03-20:
@bridadan 
````
--grm target:client:url -> TypeError
--grm target:client:url:port -> no error
````

bridadan, 2019-03-20:
Hi @alekla01,

I have a test case specifically for this case:

- Using raw IP address: https://github.com/ARMmbed/mbed-os-tools/commit/36610815fd4425ed3162057411f8b578216c5faf#diff-5f374df9fc07e86f5e318ab930107fb1R607
- Using a URL with a protocol: https://github.com/ARMmbed/mbed-os-tools/commit/36610815fd4425ed3162057411f8b578216c5faf#diff-5f374df9fc07e86f5e318ab930107fb1R615

Could you please provide the following so I can accurately reproduce the error:

- The output of `pip list` (looking for the versions of various python packages on your system
- The full command that you are executing. (specifically interested in the **unmodified** `--grm` argument. If you can't share it publicly you can always email me internally, thanks)

PR #158 (2019-04-11): Fix how missing ports are handled for grm
### Description

Fixes #141. Thanks for the help @alekla01 , I was able to patch my branch and it looks like everything works.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@alekla01
closed 2019-04-11 by bridadan