ARMmbed/mbed-os-tools #149: Missing instructions on using plugins in mbed-host-tests

MarceloSalazar, 2019-03-26:
### Description

** Customer request**

It's not clear how to use the plugins included in the mbed-host-tests here:
https://github.com/ARMmbed/mbed-os-tools/tree/master/packages/mbed-host-tests/mbed_host_tests/host_tests_plugins

Docs are very thin and aren't really useful.
https://github.com/ARMmbed/mbed-os-tools/tree/master/packages/mbed-host-tests#plugins

We need to provide basic guidance on how to set up the system to use a specific plugin (STLink preferred) and be able to run tests.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

geoffrey-vl, 2019-04-30:
Hi, how is the progress on this?

bridadan, 2019-05-01:
Currently have had any time to work on this, but I agree this documentation is missing and should be written!

I'll be honest, in general I'm not a huge fan of "plugin" architectures. In my experience they have a tendency to accumulate bugs over time (especially since they are usually optional or have optional dependencies). Do you have any suggestions to improve what we currently have?

geoffrey-vl, 2019-05-01:
Well my goal is to make greentea work with our curstom device.
We went through this previously [here](https://github.com/ARMmbed/mbed-ls/issues/414).
More recently however we were informed that greentea should work using the plugins, so here we are :)

Regarding your questions.
How that is implemented in mbed tooling doesn't really matter for us customers, I'd guess.
In the end most of us want a out-of-the-box experience, or alternatively a good manual.
What I don't understand now: should we script something ourselves to make the plugins work? Or would the correct command line syntax suffice?
Given the fact that the plugins come within the same repo I'd guess maintenance shouldn't be that problematic. It's just that it includes extra testing, you need to verify each plugin before shipping. The good thing about a plugin architecture is that people can extend it themselves, given that some guidelines are provided on how to do that. Also, code doesn't just rot because it uses a plugin architecture. One example that comes to my mind would be gstreamer. Code rots because it's left unmaintained. If you aren't planning to expose the plugin architecture to the outside world and encourage others to plugin on the tooling then I'd say chose any architecture that suits you, without giving in on functionality that is :)

As far as documentation goes, its always helpful to have a good command  line help menu which exposes all the options, also for accessing plugins. After upgrading mbedhtrun I can now see in the command line help menu the --plugins option + some info on what it does and if it is supported on my system. So that's ok. If I'm not mistaken this is what I'm looking for:

`mbedhtrun -c stlink -f /path/to/file/binary.bin`

I'd need to verify this on my Windows machine as stlink with linux is not supported currently.

geoffrey-vl, 2019-05-09:
Hi, bridadan.

That seems to be working well!

```
C:\work-mbed\myfirmware\BUILD\myhw\GCC_ARM-DEBUG>mbedhtrun -c stlink -f myfirmware.bin -p COM14
[1557407884.40][HTST][INF] host test executor ver. 0.0.8
[1557407884.40][HTST][INF] copy image onto target...
STM32 ST-LINK CLI v3.2.0.0
STM32 ST-LINK Command Line Interface

ST-LINK SN : 37FF6D064D4E383510261143
ST-LINK Firmware version : V2J29S7
Connected via SWD.
SWD Frequency = 4000K.
Target voltage = 3.2 V.
Connection mode : Normal.
Device ID:0x419
Device flash Size : 2048 Kbytes
Device family :STM32F42xxx/F43xxx
Loading file...
Flash Programming:
  File : myfirmware.bin
  Address : 0x08000000
Memory programming...
██████████████████████████████████████████████████ 100%
Memory programmed in 31s and 438ms.
Verification...OK
Programming Complete.
Programmed memory Checksum: 0x06FA02ED

[1557407920.42][MBED][WRN] Target ID not found: Skipping flash check and retry
[1557407920.44][HTST][INF] starting host test process...
[1557407921.35][CONN][INF] starting connection process...
[1557407921.35][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1557407921.35][CONN][INF] initializing serial port listener...
[1557407921.36][SERI][INF] serial(port=COM14, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1557407921.37][HTST][INF] setting timeout to: 60 sec
[1557407921.39][SERI][INF] reset device using 'default' plugin...
[1557407921.64][SERI][INF] waiting 1.00 sec after reset
[1557407922.64][SERI][INF] wait for it...
[1557407922.64][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
...
```

It does have some problems though when the device is in standby mode but I believe that's something we can deal with.

closed 2019-05-09 by bridadan