ARMmbed/mbed-os-tools #152: Bumping up DAPLink interfaces will/may happen in the near future

brianesquilona, 2019-03-27:
### Description

### Issue request type

    [ ] Question
    [x] Enhancement
    [x] Bug

[https://github.com/ARMmbed/mbed-os-tools/blob/master/src/mbed_os_tools/detect/windows.py#L36](https://github.com/ARMmbed/mbed-os-tools/blob/master/src/mbed_os_tools/detect/windows.py#L36) needs future proofing.

loverdeg-ep, 2019-03-28:
@brianesquilona What is the issue?

bridadan, 2019-03-28:
@loverdeg-ep we looked at this offline, but basically @brianesquilona was experimenting with the composite USB device of the firmware and found that Mbed LS would no longer detect the mount point for the device. I'm pretty sure this is the reason for his particular issue (I asked him to put up an issue so we didn't lose track of it).

loverdeg-ep, 2019-03-28:
My comment should have been more descriptive.
I'm a fan of most things DAPLink and was just curious what "bumping up" meant.

I checked the linked line of code and assumed increasing the number of devices within the composite but was lost as to the context of the issue.

bridadan, 2019-05-02:
@brianesquilona told me offline that this change will no longer be needed, so closing this now.

closed 2019-05-02 by bridadan