ARMmbed/mbed-os-tools #156: arm-none-eabi-g++: error: CreateProcess: No such file or directory while running through greentea

UrmilaKL, 2019-04-04:
Description: [CYW943012P6EVB_01]multiADV: arm-none-eabi-g++: error: CreateProcess: No such file or directory while running through greentea
Steps to Reproduce:
1. git clone http://git-ore.aus.cypress.com/repo/mbed-os
2. cd mbed-os
3. git checkout remotes/origin/cy-mbed-os-5.12.0-august-home
4. Run Run greentea testcases from below path
TESTS\Group1\multiADV
Expected Result: All the test cases should execute properly without any error
Actual Result: Execution is failing with the compiler error 
"[DEBUG] Output: arm-none-eabi-g++: error: CreateProcess: No such file or directory"
Frequency (1/1, 1/50, 1/100):1/1
Platform:43012
OS Version:Win-10 64bits
[Note: Compilation works fine with below command, Manually if we execute testcase by copying .hex file on to device application works properly
mbed compile --target CYW943012P6EVB_01 --toolchain GCC_ARM]
[greenteaerror.txt](https://github.com/ARMmbed/mbed-os-tools/files/3044771/greenteaerror.txt)

UrmilaKL, 2019-04-04:
Help on this issue is much appreciated..

bridadan, 2019-04-05:
Hi @UrmilaKL, this issue is most likely the same as this: https://github.com/ARMmbed/mbed-os/issues/7129

For step 4, the step you gave was:
>Run Run greentea testcases from below path
>TESTS\Group1\multiADV

Could you clarify the following:

- What comand did you run?
- What path did you run this command in?

bridadan, 2019-04-05:
Oops, I see you provided this information in your attached logs, thanks.

The only thing I can recommend you do is try to shorten any paths within the test cases or the target port. The issue https://github.com/ARMmbed/mbed-os/issues/7129 has a bit more details.

UrmilaKL, 2019-04-08:
Hi @bridadan ,

I got confused with the path here please do correct me,

shorten any paths within the test cases or the target port -> i am using below tree structure to run greentea testcases 
mbed-os -> TSETS -> Group1 -> multiADV -> main.cpp(testcase)

Should i rename all my testcases or should i change the includes file manually(includes.txt )?

When i rename my multiADV as 'm' and try to run the testcase i did not get CreateProcess error but i got below error 

"invalid test case names (specified with '-n' option)"
Please find the attached file 
[GreenTeaLogs.txt](https://github.com/ARMmbed/mbed-os-tools/files/3054102/GreenTeaLogs.txt)

UrmilaKL, 2019-04-08:
Also please find directory  Group1 folder inside TESTS folder
[Group1.zip](https://github.com/ARMmbed/mbed-os-tools/files/3054109/Group1.zip)

UrmilaKL, 2019-04-09:
@bridadan , 
Shall we copy entire TESTS folder inside c:\and run the test cases?
Will that work in that way?

UrmilaKL, 2019-04-09:
HI @bridadan ,
When i try to shorten the test case name path and try to run mbed test command using below command i am getting below error

command used: mbed test -m CYW943012P6EVB_01 -t GCC_ARM -n Y: -vv  (where Y: has tests-Group1-multiADV)
[error.txt](https://github.com/ARMmbed/mbed-os-tools/files/3060528/error.txt)

bridadan, 2019-04-09:
In your log the command is this: `mbed test -m CYW943012P6EVB_01 -t GCC_ARM -n "C:\mbed-examples\mbed-os\TESTS\mbed_hal\flash\functional_tests\main.cpp" -vv`

For the `-n` command you need to specify the test name, not the path. In this case, the command would be: `mbed test -m CYW943012P6EVB_01 -t GCC_ARM -n tests-mbed_hal-flash-functional_tests -vv`

bridadan, 2019-04-18:
Closing this due to inactivity.

closed 2019-04-18 by bridadan