ARMmbed/mbed-os-tools #167: pip install fails for v0.0.7

korjaa, 2019-05-03:
### Description
mbed-os-tools pip install fails for v0.0.7

Steps to reproduce:
```
$ virtualenv venv
Running virtualenv with interpreter /usr/bin/python2
New python executable in /home/jaakor01/workspace/temp/venv/bin/python2
Also creating executable in /home/jaakor01/workspace/temp/venv/bin/python
Installing setuptools, pkg_resources, pip, wheel...done.
```
```
$ souce venv/bin/activate
```
```
$ pip install mbed-os-tools==v0.0.7
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting mbed-os-tools
  Using cached https://files.pythonhosted.org/packages/c0/74/6a54fd817e26f754da849e6c6701ac5fc0039ebc2a4624b88548b0b519a9/mbed-os-tools-0.0.7.tar.gz
    ERROR: Complete output from command python setup.py egg_info:
    ERROR: Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-3LnNys/mbed-os-tools/setup.py", line 60, in <module>
        tests_require=read("test_requirements.txt").splitlines(),
      File "/tmp/pip-install-3LnNys/mbed-os-tools/setup.py", line 27, in read
        with open(os.path.join(os.path.dirname(__file__), fname), 'r') as f:
    IOError: [Errno 2] No such file or directory: '/tmp/pip-install-3LnNys/mbed-os-tools/test_requirements.txt'
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-3LnNys/mbed-os-tools/
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

rutgervandenberg, 2019-05-03:
I'm having the same problem. This is breaking my CI builds right now.

korjaa, 2019-05-03:
For me this bug was breaking `icetea` installation. I was able to workaround that by installing old-version of `mbed-ls==1.7.2` beforehand.

For example below worked for me:
```
$ virtualenv venv
$ source venv/bin/activate
$ pip install mbed-ls==1.7.2
$ pip install icetea
```

Without the `mbed-ls==1.7.2`, `icetea` would pull the latest `mbed-os-tools` from some route.

korjaa, 2019-05-03:
Proposed fix https://github.com/ARMmbed/mbed-os-tools/pull/168

0xc0170, 2019-05-03:
Thanks @korjaa for looking at this. Also affecting our CI at the moment, will review the fix

jonikula, 2019-05-03:
I just observed the same issue when installing mbed-flasher. @korjaa Icetea depends on Mbed-flasher which depends on mbed-ls which now depends on this package.

cmonr, 2019-05-03:
It moved because `pip install` does better dependency management than simply invoking `setup.py`.

bridadan, 2019-05-03:
Sorry about this folks, new release coming soon with a fix.

bridadan, 2019-05-03:
Ok, new release just went out. It fixed the pip installation for me. Let me know if you have further issues, sorry again for the breakage! I will be getting an extra check into CI to make sure this failure doesn't happen again.

closed 2019-05-03 by bridadan

korjaa, 2019-05-06:
Yep, `icetea` installation works again with this new release v0.0.8.