ARMmbed/mbed-os-tools #175: mbed test linker fails when using --build option due to duplicate main.cpp

geoffrey-vl, 2019-05-14:
### Description

- mbed-os 5.9.2
- custom board based upon STM32 F429ZI
- mbed test
- gcc_arm 7.2.1

On Windows we're stuck to the maximum command length of 32k.
To bypass the `mbed test` compile errors we use the `--build` option.
Unfortunately the linker has some problems with duplicate main.cpp found.

For example when we try to execute the sd-driver tests we use following command:

```
C:\work-mbed\myproject>mbed test -m MY_DEVICE -t GCC_ARM -n sd-driver-tests-block_device-basic --profile=profiles/release11.json --compile --build "C:/BUILD"
```

Here is the output:
```
[mbed] Working path "C:\work-mbed\myproject" (program)
Scan: .
Scan: FEATURE_LWIP
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Scan: TESTS
Building library mbed-build (MY_DEVICE , GCC_ARM)
Scan: .
Scan: FEATURE_LWIP
Compile [100.0%]: main.cpp
Scan: BUILD
Scan: FEATURE_LWIP
[Error] @,: Compiler version mismatch: Have 7.2.1; expected version >= 6.0.0 and < 7.0.0
Building project basic (MY_DEVICE , GCC_ARM)
Scan: BUILD
Scan: basic
Scan: FEATURE_LWIP
Link: basic
../../BUILD/sd-driver/TESTS/block_device/basic/basic.o: In function `main':
C:\work-mbed\myproject/.\sd-driver\TESTS\block_device\basic/basic.cpp:177: multiple definition of `main'
../../BUILD/main.o:C:\work-mbed\myproject/.\main.cpp:16: first defined here
collect2.exe: error: ld returned 1 exit status
Memory map breakdown for built projects (values in Bytes):
+------+--------+-----------+------------+-------------+
| name | target | toolchain | static_ram | total_flash |
+------+--------+-----------+------------+-------------+
+------+--------+-----------+------------+-------------+

Build successes:
  * MY_DEVICE ::GCC_ARM::MBED-BUILD

Build failures:
  * MY_DEVICE ::GCC_ARM::SD-DRIVER-TESTS-BLOCK_DEVICE-BASIC
        [Error] @,: Compiler version mismatch: Have 7.2.1; expected version >= 6.0.0 and < 7.0.0
        Building project basic (MY_DEVICE , GCC_ARM)
        Scan: BUILD
        Scan: basic
        Scan: FEATURE_LWIP
        Link: basic
        ../../BUILD/sd-driver/TESTS/block_device/basic/basic.o: In function `main':
        C:\work-mbed\myproject/.\sd-driver\TESTS\block_device\basic/basic.cpp:177: multiple definition of `main'
        ../../BUILD/main.o:C:\work-mbed\myproject/.\main.cpp:16: first defined here
        collect2.exe: error: ld returned 1 exit status

[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\work-mbed\myproject"
       Command: "c:\python27\python.exe -u C:\work-mbed\myproject\mbed-os\tools\test.py -D MBED_TEST_MODE --profile profiles/release11.json -t GCC_ARM -m MY_DEVICE --source . --build C:/BUILD --test-spec C:/BUILD\test_spec.json -n sd-driver-tests-block_device-basic"
       Tip: You could retry the last command with "-v" flag for verbose output
---

```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

linlingao, 2019-05-14:
You'll need to remove main.cpp in the root directory.
See Note on this page https://os.mbed.com/docs/mbed-os/v5.12/tools/test-and-debug.html
Note: mbed test does not work in applications that contain a main function that is outside of a TESTS directory.

geoffrey-vl, 2019-05-14:
Okay, but is also states `# Optional main.cpp with main() if it is an application module`.
Could you elaborate a bit more? Because it is a bit weird for me to have my testing farm pull my application repository and next run a script that delete files in that repo.

linlingao, 2019-05-14:
That segment shows typical code directory structure. For example, when you import a program such as blinky, you'll find a main.cpp in the root directory. You also get TESTS as well but "mbed compile" knows the code under TESTS needs to be excluded.
Now if you want to run "mbed test", it picks up the main.cpp in the TESTS directory, as well as "main.cpp" under the root directory. You just can't have two main.cpp files. Therefore the main.cpp at the root directory has to be removed/renamed.
One possible enhancement might be, ignore main.cpp at the root directory when "mbed test" is invoked. I would have to check with our tools team to see how much effort it requires. In the mean time, I think you'll have to run a script to rename that main.cpp to say main.txt during test.

geoffrey-vl, 2019-05-15:
I can confirm that simply removing the application's `main.cpp` is indeed working.
I've also tried adding a `.mbed` file with content `ROOT=.` within the to be tested library, but `mbed test` failed in finding its tools.
One other way I can think of solving this issue is to create an empty application that contains all of the libraries that we want to test, but that would require extra maintenance and we wouldn't be able to test our application code.

As a result, `mbed test` can't be used "out of the box" for testing application code and non mbed-os libraries (for example sd-driver from within an application). I'd be interested in an enhancement, I think it would also improve the user experience in general. Your solution could work for us. One other possible solution would be of we could have something like `.mbedignore` work specifically for testing, i.e. something like `.mbedtestignore`. That would add flexibility for developers to choose their preferred directory structure. What do you think?

geoffrey-vl, 2019-05-15:
@linlingao 
I've just noticed in the latest mbed-os documentation that a special compile guard was added specifically for `mbed test` : 

```
#if !MBED_TEST_MODE
int main() {
    // Application code
}
#endif
```

After upgrading our toolchain and adding that compile guard to our application I was able to perform the greentea tests without any linking errors occurring. So I think we have a solution there, I'm closing this one, feel free to reopen if you'd like anything to be added.

closed 2019-05-15 by geoffrey-vl