ARMmbed/mbed-os-tools #179: KeyError in test_api.py

Alex-EEE, 2019-05-16:
### Description

Getting a strange KeyError from test_api.py when running mbed test compile with the release.json profile.  The exact same code doesn't do this under debug.json profile.  

-Any thoughts on what could be going on?  
-Could I safely put this line of code under a try... seems to just be putting together some kind of report? 

```
2019-05-16T21:42:18.8664526Z | Module                               |           .text |       .data |          .bss |
 
2019-05-16T21:42:18.8664526Z |--------------------------------------|-----------------|-------------|---------------|
 
2019-05-16T21:42:18.8664526Z | IAR-RELEASE\features                 |   14514(+14514) |   317(+317) |       40(+40) |
 
2019-05-16T21:42:18.8664526Z | IAR-RELEASE\mbed                     | 193379(+193379) | 1030(+1030) | 42316(+42316) |
 
2019-05-16T21:42:18.8664526Z | IAR-RELEASE\test-server-dependencies |   26286(+26286) |     20(+20) |       28(+28) |
 
2019-05-16T21:42:18.8664526Z | [lib]\dl7M_tlf.a                     |   22708(+22708) |   408(+408) |     980(+980) |
 
2019-05-16T21:42:18.8664526Z | [lib]\dlpp7M_tl_fc.a                 |       855(+855) |     16(+16) |       24(+24) |
 
2019-05-16T21:42:18.8664526Z | [lib]\m7M_tls.a                      |     5136(+5136) |       0(+0) |         0(+0) |
 
2019-05-16T21:42:18.8664526Z | [lib]\rt7M_tl.a                      |     1032(+1032) |       0(+0) |         0(+0) |
 
2019-05-16T21:42:18.8820761Z | [lib]\th7M_tlf.a                     |       276(+276) |       0(+0) |       16(+16) |
 
2019-05-16T21:42:18.8820761Z | [misc]                               |       641(+641) |       0(+0) |     384(+384) |
 
2019-05-16T21:42:18.8820761Z | features\AiLwm2mProcedure            |     3732(+3732) |       0(+0) |   1960(+1960) |
 
2019-05-16T21:42:18.8820761Z | Subtotals                            | 268559(+268559) | 1791(+1791) | 45748(+45748) |
 
2019-05-16T21:42:18.8820761Z Total Static RAM memory (data + bss): 47539(+47539) bytes
 
2019-05-16T21:42:18.8820761Z Total Flash memory (text + data): 270350(+270350) bytes
 
2019-05-16T21:42:18.8820761Z Image: BUILD/tests/K64F/features/AiLwm2mProcedure/TESTS/group_1/case_1/case_1.bin
 
2019-05-16T21:42:19.5539186Z Traceback (most recent call last):
 
2019-05-16T21:42:19.5539186Z   File "C:\firmware\tfs-agent\_work\6\s\core\mbed\mbed-os\tools\test.py", line 277, in <module>
 
2019-05-16T21:42:19.5539186Z     spe_build=mcu_secured)
 
2019-05-16T21:42:19.5539186Z   File "C:\firmware\tfs-agent\_work\6\s\core\mbed\mbed-os\tools\test_api.py", line 2341, in build_tests
 
2019-05-16T21:42:19.5539186Z     report_entry[worker_result['kwargs']['project_id'].upper()][0][0]['output'] = new_notify.get_output()
 
2019-05-16T21:42:19.5539186Z KeyError: 'MBED-MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_ASSERTION_FAILURE_TEST_SETUP'
 
2019-05-16T21:42:19.5539186Z [ERROR] 'MBED-MBED-OS-FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_ASSERTION_FAILURE_TEST_SETUP'
 
2019-05-16T21:42:19.6164156Z [mbed] Working path "C:\firmware\tfs-agent\_work\6\s\core\mbed" (library)
 
2019-05-16T21:42:19.6164156Z [mbed] Program path "C:\firmware\tfs-agent\_work\6\s\core\mbed"
 
2019-05-16T21:42:23.5068802Z ##[error][mbed] ERROR: "c:\python27\python.exe" returned error.
 
2019-05-16T21:42:23.5068802Z ##[error]       Code: 1
 
2019-05-16T21:42:23.5068802Z ##[error]       Path: "C:\firmware\tfs-agent\_work\6\s\core\mbed"
 
2019-05-16T21:42:23.5068802Z ##[error]       Command: "c:\python27\python.exe -u C:\firmware\tfs-agent\_work\6\s\core\mbed\mbed-os\tools\test.py -D MBED_TEST_MODE --profile mbed-os/tools/profiles/release.json -t IAR -m K64F -c --source .. --build .\BUILD\tests\K64F\IAR-RELEASE --test-spec .\BUILD\tests\K64F\IAR-RELEASE\test_spec.json --build-data .\BUILD\tests\K64F\IAR-RELEASE\build_data.json --app-config ../test-server-dependencies/k64/mbed_app.json --greentea --paths .."
 
2019-05-16T21:42:23.5068802Z ##[error]       Tip: You could retry the last command with "-v" flag for verbose output
 
2019-05-16T21:42:23.5068802Z ##[error]---
 
2019-05-16T21:42:23.5068802Z ##[error]Process completed with exit code 1 and had 1 error(s) written to the error stream.
```
### Issue request type

    [ x] Question
    [ ] Enhancement
    [ ] Bug

mark-edgeworth, 2019-05-22:
Hi, can you let me know what version of the tools you are using please?

Alex-EEE, 2019-05-22:
Commit 5c50c9f of mbed-os-tools, alongside mbed-os 12.2.

I didn't realize there had been so much activity in the mbed-os-tools repo, I'll try updating to v0.0.8 and see if that makes a difference

Alex-EEE, 2019-06-19:
Ok, I updated to latest here (b048) and still had the issue.  We figured out the root cause:
If you call mbed test --compile with two different --profile ( say, debug and release ), and you don't change the --build directory, whichever compile runs second will fail with this error.

The workaround is to change build directories between calls to mbed test --compile with different --profile

mark-edgeworth, 2019-06-20:
Ah, ok, that'll do it. Many thanks for the info - I'll see if we can improve this at some point. Are you happy to close this issue?

closed 2019-10-31 by ciarmcom