ARMmbed/mbed-os-tools #182: Use the same mbed target Id for different mbed-os target

jeromecoutant, 2019-06-13:
### Description

I would like to have different HW package using the same mbed target Id.

In mbedls, database is configuring each target:
- u"1234": u"MY_TARGET",

In targets.json mbed-os file, the detect_code macro can be available and is no more used.

    "MY_TARGET": {
        "detect_code": ["1234"],

Proposition is to re-use it, and to be able to create a new target using the same detect code Id.

    "MY_NEW_TARGET": {
        "detect_code": ["1234"],

mbed test command could work for both targets:

mbed test -m MY_TARGET
	will work as usual

mbed test -m MY_NEW_TARGET
	will work, even if MY_NEW_TARGET is not in mbedls database

Maybe it could be easier to use directly the target name to detect:

    "MY_NEW_TARGET": {
        "detect_code": ["MY_TARGET"],

@MarceloSalazar 
@LMESTM 

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

MarceloSalazar, 2019-06-13:
@jeromecoutant thanks for raising this request. I agree with you and believe it's important to have this place to let us introduce different variances of platforms that may require a slightly different configuration.

FYI @ARMmbed/mbed-os-maintainers

jeromecoutant, 2019-06-19:
Any chance to get this update soon,
or it is better to find a workaround ? :-)

mark-edgeworth, 2019-06-19:
Hi @jeromecoutant, I'm sorry but it is unlikely we'll get to this anytime soon.

mark-edgeworth, 2019-09-16:
Just for the record: the target names are currently restricted to a maximum of 20 characters...

closed 2019-10-31 by ciarmcom