ARMmbed/mbed-os-tools #191: How to use python 3 for mbed CLI

andrewc-arm, 2019-08-28:
### Description

Hi,
I am using Windows and I wish to change mbed CLI to use Python 3.
It's because Python 3 supports long path in Windows.
https://stackoverflow.com/questions/46488118/win32-long-paths-from-python
There are multiple issues with Windows + mbed on path limitation.
How do I upgrade mbed CLI to use Python 3?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

madchutney, 2019-09-04:
You should be able to run mbed-cli under which ever python interpreter you have installed (Python >=2.7.10, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.0, !=3.4.1, !=3.4.2, <4).

Although I don't think swapping to Python 3 will fix the long the long path issue.

andrewc-arm, 2019-09-05:
Thanks, @madchutney .
I will simply change the system's Python version and retry.

closed 2019-09-05 by andrewc-arm

madchutney, 2019-09-05:
@andrewc-arm I recommend that you check out `pipenv`, don't change your system version of Python.