ARMmbed/mbed-os-tools #201: MbedStudio / target mocking

pfried, 2019-10-14:
### Description

The mbedls library provides a way to add custom targets by overriding an existing target (mock). This works also by putting a `mbedls.json` file in the `CWD` of the library.

As I guess MbedStudio also uses the library to do the auto target detection I wondered if it was possible to fake a custom target to MbedStudio? If the above mechanism works, where to put the file?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

madchutney, 2019-12-04:
Hi Friedrich,

Can you please ask questions on our forum site, where the appropriate team will be able to answer your question:
- https://forums.mbed.com/c/More-on-our-next-gen-desktop-IDE

Thanks,
Graham

closed 2019-12-04 by madchutney

pfried, 2019-12-06:
Hi Graham,

I did, so for reference: https://forums.mbed.com/t/custom-targets-debugging-and-auto-detection/6177

Best, Friedrich