ARMmbed/mbed-os-tools #205: GreenTea text report crash with python 3

jeromecoutant, 2019-11-08:
### Description

Done command:
mbedgt -t DISCO_F746NG-IAR --report-text=report.txt

It's working on a PC with python 2.7 installed,
but not with a PC with python 3.7:

mbedgt: exporting to TEXT 'report.txt'...
mbedgt: unexpected error:
        'str' object has no attribute 'decode'
AttributeError: 'str' object has no attribute 'decode'

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

VeliMattiLahtela, 2019-11-14:
@jeromecoutant python3 reporting was fixed for mbedgt v1.7.3, could you try with that

jamesbeyond, 2019-12-04:
Currently, 1.7.3 greentea on Pypi is not support python3,
the fix will be in the next greentea release, which will happen soon,
The next version will support python 3

Workaround:
pip3 install git+https://github.com/ARMmbed/mbed-os-tools.git@master#subdirectory=packages/mbed-greentea/

jamesbeyond, 2019-12-06:
Hi @jeromecoutant,
we having a new release of mbed-greentea 1.7.4, https://pypi.org/project/mbed-greentea/
would you able to confirm this issue is fixed, and closing the issue?

jamesbeyond, 2020-03-16:
We think this issue been fixed. please re-open the ticket if you still experience this issue

closed 2020-03-16 by jamesbeyond