ARMmbed/mbed-os-tools #217: mbed CLI (1.10.2) is unable to detect S5JS100

andrewc-arm, 2020-02-14:
### Description

Hi, 
mbed CLI (1.10.2) is unable to detect S5JS100.
S5JS100 is registered here.
https://github.com/ARMmbed/mbed-os-tools/blob/master/src/mbed_os_tools/detect/platform_database.py#L293
However, my mbed CLI (just upgraded via `pip install --upgrade mbed-cli`) fails to detect S5JS100 like this.
```
andcho01@a030002-lin:~/code/greentea-S111-0$ mbedls
WARNING:mbedls.lstools_base:daplink entry: "3701" not found in platform database
| platform_name | platform_name_unique | mount_point   | serial_port  | target_id                                        | daplink_version |
|---------------|----------------------|---------------|--------------|--------------------------------------------------|-----------------|
| unknown       | None[0]              | /media/MBED_B | /dev/ttyACM0 | 3701000008cd49dc00000000000000000000000097969902 | 0247            |
andcho01@a030002-lin:~/code/greentea-S111-0$ mbed --version
1.10.2
```

Could you kindly guide me on how I can detect without `mbedls -mock`?

Thank you,
### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

toyowata, 2020-03-04:
@andrewc-arm 
I had similar experience using another board.  You can try this...

```
pip uninstall mbed-os-tools
pip install mbed-os-tools
```

andrewc-arm, 2020-03-04:
@toyowata 
Thanks for the tip! I will try that later and feedback to this issue. 👍

closed 2020-03-04 by andrewc-arm