ARMmbed/mbed-os-tools #231: Mbed-Greentea: --shuffle flag does not work with Python3

dustin-crossman, 2020-06-30:
### Description

Passing the --shuffle flag to mbedgt as in the following:
mbed test -t GCC_ARM -m CY8CPROTO_062_4343W -n "\*mbed_hal-\*" --compile
mbedgt --shuffle
results in the following error:

```
Traceback (most recent call last):
  File "C:\Users\dcro\AppData\Local\Programs\Python\Python37\Scripts\mbedgt-script.py", line 11, in <module>
    load_entry_point('mbed-greentea==1.7.4', 'console_scripts', 'mbedgt')()
  File "c:\users\dcro\appdata\local\programs\python\python37\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 358, in main
    cli_ret = main_cli(opts, args)
  File "c:\users\dcro\appdata\local\programs\python\python37\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 872, in main_c
li
    random.shuffle(filtered_ctest_test_list_keys, lambda: shuffle_random_seed)
  File "c:\users\dcro\appdata\local\programs\python\python37\lib\random.py", line 284, in shuffle
    x[i], x[j] = x[j], x[i]
TypeError: 'dict_keys' object is not subscriptable
```

The issue is the .keys() call here: 
https://github.com/ARMmbed/mbed-os-tools/blob/master/packages/mbed-greentea/mbed_greentea/mbed_greentea_cli.py#L869
In python2 .keys() returns a list of the dictionary keys.
In python3 .keys() returns a view object (dict_keys) that is not subscriptable.

Explicitly converting filtered_ctest_test_list_keys to a list should be all that is required to fix the issue for both py2 and py3.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

rwalton-arm, 2020-06-30:
Quick fix would be `filtered_ctest_test_list_keys = list(filtered_ctest_test_list.keys())`. In python 3 `dict.keys` returns a `keysview` and not a `list`.

dustin-crossman, 2020-06-30:
Just tested that as a fix and it works well. I can put in a PR if that fix seems sufficient to resolve this.