ARMmbed/mbed-os-tools #245: Error while running greentea on DISCO_L475VG_IOT01A

Nakul93, 2020-11-09:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug
Unable to run even a simple test of greentea in DISCO_L475VG_IOT01A board, tried it with NUCLEO F091RC board and that works. Following is the log that I've obtained,
Building project simple-test (DISCO_L475VG_IOT01A, GCC_ARM)
Scan: GCC_ARM
Scan: simple-test
Macros: -DDEVICE_USTICKER=1 -DDEVICE_MPU=1 -DDEVICE_WATCHDOG=1 -DARM_MATH_CM4 -DTARGET_CORTEX_M -D__FPU_PRESENT=1 -D__MBED_CMSIS_RTOS_CM -DMBED_BUILD_TIMESTAMP=1604921936.294072 -DFEATURE_BLE=1 -DCOMPONENT_FLASHIAP=1 -DDEVICE_ANALOGIN=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_M4 -DTARGET_DISCO_L475VG_IOT01A -DTOOLCHAIN_GCC -DDEVICE_LPTICKER=1 -DDEVICE_RESET_REASON=1 -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -DTARGET_STM -DDEVICE_STDIO_MESSAGES=1 -DTARGET_LIKE_MBED -DTARGET_NAME=DISCO_L475VG_IOT01A -DDEVICE_TRNG=1 -DDEVICE_I2C_ASYNCH=1 -DMBED_TEST_MODE -DTARGET_RTOS_M4_M7 -DTARGET_CORTEX -DTARGET_RELEASE -DDEVICE_I2CSLAVE=1 -DCOMPONENT_BlueNRG_MS=1 -D__CORTEX_M4 -DTARGET_FF_ARDUINO -DDEVICE_FLASH=1 -DDEVICE_QSPI=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DDEVICE_SLEEP=1 -DDEVICE_CRC=1 -DCOMPONENT_QSPIF=1 -D__MBED__=1 -DDEVICE_SPI=1 -DUSE_HAL_DRIVER -DSTM32L475xx -DDEVICE_I2C=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_SERIAL=1 -DTOOLCHAIN_GCC_ARM -DTARGET_MCU_STM32L475xG -DDEVICE_RTC=1 -DTARGET_MCU_STM32L4 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_PWMOUT=1 -DDEVICE_ANALOGOUT=1 -DDEVICE_CAN=1 -DDEVICE_USBDEVICE=1 -DTARGET_MX25R6435F -DMBED_TICKLESS -DTARGET_MCU_STM32 -DTARGET_STM32L4 -DEXTRA_IDLE_STACK_REQUIRED -DDEVICE_PORTIN=1 -DUSE_FULL_LL_DRIVER -DDEVICE_SPISLAVE=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_CORDIO -DMBED_SPLIT_HEAP -DTARGET_STM32L475xG
Compile [100.0%]: main.cpp
Compile: C:\Program Files (x86)\GNU Arm Embedded Toolchain\9 2020-q2-update\bin\arm-none-eabi-g++ -c -std=gnu++14 -fno-rtti -Wvla -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -MMD -fomit-frame-pointer -Os -g -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_MINIMAL_PRINTF -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x100000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x18000 -DMBED_RAM1_START=0x10000000 -DMBED_RAM1_SIZE=0x8000 -DDEVICE_USTICKER=1 -DDEVICE_MPU=1 -DDEVICE_WATCHDOG=1 -DARM_MATH_CM4 -DTARGET_CORTEX_M -D__FPU_PRESENT=1 -D__MBED_CMSIS_RTOS_CM -DMBED_BUILD_TIMESTAMP=1604921936.294072 -DFEATURE_BLE=1 -DCOMPONENT_FLASHIAP=1 -DDEVICE_ANALOGIN=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_M4 -DTARGET_DISCO_L475VG_IOT01A -DTOOLCHAIN_GCC -DDEVICE_LPTICKER=1 -DDEVICE_RESET_REASON=1 -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -DTARGET_STM -DDEVICE_STDIO_MESSAGES=1 -DTARGET_LIKE_MBED -DTARGET_NAME=DISCO_L475VG_IOT01A -DDEVICE_TRNG=1 -DDEVICE_I2C_ASYNCH=1 -DMBED_TEST_MODE -DTARGET_RTOS_M4_M7 -DTARGET_CORTEX -DTARGET_RELEASE -DDEVICE_I2CSLAVE=1 -DCOMPONENT_BlueNRG_MS=1 -D__CORTEX_M4 -DTARGET_FF_ARDUINO -DDEVICE_FLASH=1 -DDEVICE_QSPI=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DDEVICE_SLEEP=1 -DDEVICE_CRC=1 -DCOMPONENT_QSPIF=1 -D__MBED__=1 -DDEVICE_SPI=1 -DUSE_HAL_DRIVER -DSTM32L475xx -DDEVICE_I2C=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_SERIAL=1 -DTOOLCHAIN_GCC_ARM -DTARGET_MCU_STM32L475xG -DDEVICE_RTC=1 -DTARGET_MCU_STM32L4 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_PWMOUT=1 -DDEVICE_ANALOGOUT=1 -DDEVICE_CAN=1 -DDEVICE_USBDEVICE=1 -DTARGET_MX25R6435F -DMBED_TICKLESS -DTARGET_MCU_STM32 -DTARGET_STM32L4 -DEXTRA_IDLE_STACK_REQUIRED -DDEVICE_PORTIN=1 -DUSE_FULL_LL_DRIVER -DDEVICE_SPISLAVE=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_CORDIO -DMBED_SPLIT_HEAP -DTARGET_STM32L475xG @.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\.\TESTS\test-group\simple-test\.includes_c3b83ba20cc27fc4bfe3c675121f383e.txt -include .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\.\TESTS\test-group\simple-test\mbed_config.h -MD -MF BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\TESTS\test-group\simple-test\TESTS\test-group\simple-test\main.d -o BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\TESTS\test-group\simple-test\TESTS\test-group\simple-test\main.o .\TESTS\test-group\simple-test\main.cpp
[DEBUG] Return: 1
[DEBUG] Output: arm-none-eabi-g++: fatal error: cannot execute 'c:/program files (x86)/gnu arm embedded toolchain/9 2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/as.exe': CreateProcess: No such file or directory
[DEBUG] Output: compilation terminated.

Memory map breakdown for built projects (values in Bytes):
| name | target | toolchain | static_ram | total_flash |
|------|--------|-----------|------------|-------------|

Build successes:
  * DISCO_L475VG_IOT01A::GCC_ARM::MBED-BUILD

Build failures:
  * DISCO_L475VG_IOT01A::GCC_ARM::TESTS-TEST-GROUP-SIMPLE-TEST
        Building project simple-test (DISCO_L475VG_IOT01A, GCC_ARM)
        Scan: GCC_ARM
        Scan: simple-test
        Macros: -DDEVICE_USTICKER=1 -DDEVICE_MPU=1 -DDEVICE_WATCHDOG=1 -DARM_MATH_CM4 -DTARGET_CORTEX_M -D__FPU_PRESENT=1 -D__MBED_CMSIS_RTOS_CM -DMBED_BUILD_TIMESTAMP=1604921936.294072 -DFEATURE_BLE=1 -DCOMPONENT_FLASHIAP=1 -DDEVICE_ANALOGIN=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_M4 -DTARGET_DISCO_L475VG_IOT01A -DTOOLCHAIN_GCC -DDEVICE_LPTICKER=1 -DDEVICE_RESET_REASON=1 -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -DTARGET_STM -DDEVICE_STDIO_MESSAGES=1 -DTARGET_LIKE_MBED -DTARGET_NAME=DISCO_L475VG_IOT01A -DDEVICE_TRNG=1 -DDEVICE_I2C_ASYNCH=1 -DMBED_TEST_MODE -DTARGET_RTOS_M4_M7 -DTARGET_CORTEX -DTARGET_RELEASE -DDEVICE_I2CSLAVE=1 -DCOMPONENT_BlueNRG_MS=1 -D__CORTEX_M4 -DTARGET_FF_ARDUINO -DDEVICE_FLASH=1 -DDEVICE_QSPI=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DDEVICE_SLEEP=1 -DDEVICE_CRC=1 -DCOMPONENT_QSPIF=1 -D__MBED__=1 -DDEVICE_SPI=1 -DUSE_HAL_DRIVER -DSTM32L475xx -DDEVICE_I2C=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_SERIAL=1 -DTOOLCHAIN_GCC_ARM -DTARGET_MCU_STM32L475xG -DDEVICE_RTC=1 -DTARGET_MCU_STM32L4 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_PWMOUT=1 -DDEVICE_ANALOGOUT=1 -DDEVICE_CAN=1 -DDEVICE_USBDEVICE=1 -DTARGET_MX25R6435F -DMBED_TICKLESS -DTARGET_MCU_STM32 -DTARGET_STM32L4 -DEXTRA_IDLE_STACK_REQUIRED -DDEVICE_PORTIN=1 -DUSE_FULL_LL_DRIVER -DDEVICE_SPISLAVE=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_CORDIO -DMBED_SPLIT_HEAP -DTARGET_STM32L475xG
        Compile [100.0%]: main.cpp
        Compile: C:\Program Files (x86)\GNU Arm Embedded Toolchain\9 2020-q2-update\bin\arm-none-eabi-g++ -c -std=gnu++14 -fno-rtti -Wvla -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -MMD -fomit-frame-pointer -Os -g -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_MINIMAL_PRINTF -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x100000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x18000 -DMBED_RAM1_START=0x10000000 -DMBED_RAM1_SIZE=0x8000 -DDEVICE_USTICKER=1 -DDEVICE_MPU=1 -DDEVICE_WATCHDOG=1 -DARM_MATH_CM4 -DTARGET_CORTEX_M -D__FPU_PRESENT=1 -D__MBED_CMSIS_RTOS_CM -DMBED_BUILD_TIMESTAMP=1604921936.294072 -DFEATURE_BLE=1 -DCOMPONENT_FLASHIAP=1 -DDEVICE_ANALOGIN=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_M4 -DTARGET_DISCO_L475VG_IOT01A -DTOOLCHAIN_GCC -DDEVICE_LPTICKER=1 -DDEVICE_RESET_REASON=1 -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -DTARGET_STM -DDEVICE_STDIO_MESSAGES=1 -DTARGET_LIKE_MBED -DTARGET_NAME=DISCO_L475VG_IOT01A -DDEVICE_TRNG=1 -DDEVICE_I2C_ASYNCH=1 -DMBED_TEST_MODE -DTARGET_RTOS_M4_M7 -DTARGET_CORTEX -DTARGET_RELEASE -DDEVICE_I2CSLAVE=1 -DCOMPONENT_BlueNRG_MS=1 -D__CORTEX_M4 -DTARGET_FF_ARDUINO -DDEVICE_FLASH=1 -DDEVICE_QSPI=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DDEVICE_SLEEP=1 -DDEVICE_CRC=1 -DCOMPONENT_QSPIF=1 -D__MBED__=1 -DDEVICE_SPI=1 -DUSE_HAL_DRIVER -DSTM32L475xx -DDEVICE_I2C=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_SERIAL=1 -DTOOLCHAIN_GCC_ARM -DTARGET_MCU_STM32L475xG -DDEVICE_RTC=1 -DTARGET_MCU_STM32L4 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_PWMOUT=1 -DDEVICE_ANALOGOUT=1 -DDEVICE_CAN=1 -DDEVICE_USBDEVICE=1 -DTARGET_MX25R6435F -DMBED_TICKLESS -DTARGET_MCU_STM32 -DTARGET_STM32L4 -DEXTRA_IDLE_STACK_REQUIRED -DDEVICE_PORTIN=1 -DUSE_FULL_LL_DRIVER -DDEVICE_SPISLAVE=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_CORDIO -DMBED_SPLIT_HEAP -DTARGET_STM32L475xG @.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\.\TESTS\test-group\simple-test\.includes_c3b83ba20cc27fc4bfe3c675121f383e.txt -include .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\.\TESTS\test-group\simple-test\mbed_config.h -MD -MF BUILD\tests\DISC [... cut ...]
       Path: "C:\Users\naknai01\Desktop\example\New folder\mbed-os-example-blinky"
       Command: "c:\users\naknai01\appdata\local\programs\python\python37\python.exe -u C:\Users\naknai01\Desktop\example\New folder\mbed-os-example-blinky\mbed-os\tools\test.py -D MBED_TEST_MODE -t GCC_ARM -m DISCO_L475VG_IOT01A --source . --build .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM --test-spec .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json --build-data .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\build_data.json -n tests-test-group-simple-test -v --greentea"
---

jeromecoutant, 2020-11-09:
Hi

I suggest to remove cellular files in your  .mbedignore
and apply patch from https://github.com/ARMmbed/mbed-os/pull/13809

@MarceloSalazar

Nakul93, 2020-11-09:
Hi,
So when I added this .mbedignore file with
`mbed-os/connectivity/cellular*
mbed-os/connectivity/drivers/cellular*
`

I got following error while compiling,
Compile [ 65.0%]: CellularNonIPSocket.cpp
Compile: C:\Program Files (x86)\GNU Arm Embedded Toolchain\9 2020-q2-update\bin\arm-none-eabi-g++ -c -std=gnu++14 -fno-rtti -Wvla -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -MMD -fomit-frame-pointer -Os -g -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_MINIMAL_PRINTF -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x100000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x18000 -DMBED_RAM1_START=0x10000000 -DMBED_RAM1_SIZE=0x8000 -DDEVICE_I2CSLAVE=1 -DDEVICE_PORTINOUT=1 -DTARGET_MX25R6435F -DDEVICE_I2C_ASYNCH=1 -DDEVICE_WATCHDOG=1 -DDEVICE_USTICKER=1 -DMBED_TICKLESS -DDEVICE_SERIAL_FC=1 -DTOOLCHAIN_GCC -DSTM32L475xx -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_QSPI=1 -DDEVICE_SPI=1 -DDEVICE_CRC=1 -DEXTRA_IDLE_STACK_REQUIRED -DDEVICE_SERIAL=1 -DUSE_HAL_DRIVER -DTARGET_LIKE_CORTEX_M4 -DMBED_SPLIT_HEAP -DFEATURE_BLE=1 -DTARGET_STM32L4 -DTARGET_STM -DDEVICE_SERIAL_ASYNCH=1 -DTARGET_RTOS_M4_M7 -DTARGET_MCU_STM32 -DTARGET_FF_ARDUINO -DCOMPONENT_FLASHIAP=1 -DMBED_BUILD_TIMESTAMP=1604925351.1380455 -DDEVICE_ANALOGIN=1 -DDEVICE_MPU=1 -DTARGET_DISCO_L475VG_IOT01A -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_SPISLAVE=1 -DDEVICE_LPTICKER=1 -DTARGET_MCU_STM32L4 -D__CORTEX_M4 -DDEVICE_CAN=1 -DMBED_TEST_MODE -DDEVICE_I2C=1 -DTARGET_NAME=DISCO_L475VG_IOT01A -DTARGET_CORTEX -DDEVICE_INTERRUPTIN=1 -DDEVICE_FLASH=1 -DDEVICE_SPI_ASYNCH=1 -D__CMSIS_RTOS -DTARGET_CORTEX_M -DDEVICE_PORTIN=1 -DDEVICE_PWMOUT=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_M4 -DTARGET_STM32L475xG -DTARGET_LIKE_MBED -DDEVICE_USBDEVICE=1 -DDEVICE_PORTOUT=1 -DDEVICE_ANALOGOUT=1 -DARM_MATH_CM4 -DTARGET_RELEASE -DDEVICE_TRNG=1 -DCOMPONENT_QSPIF=1 -DTARGET_MCU_STM32L475xG -DDEVICE_RESET_REASON=1 -D__FPU_PRESENT=1 -DUSE_FULL_LL_DRIVER -DCOMPONENT_BlueNRG_MS=1 -DTARGET_CORDIO -D__MBED__=1 -DDEVICE_RTC=1 -D__MBED_CMSIS_RTOS_CM @.\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\.includes_e5b77d039415dd2ce8faf99adf04fb68.txt -include .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\mbed_config.h -MD -MF BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\mbed-os\connectivity\netsocket\source\CellularNonIPSocket.d -o BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\mbed-os\connectivity\netsocket\source\CellularNonIPSocket.o .\mbed-os\connectivity\netsocket\source\CellularNonIPSocket.cpp
[Fatal Error] CellularNonIPSocket.h@28,10: CellularContext.h: No such file or directory
[DEBUG] Return: 1
[DEBUG] Output: In file included from .\mbed-os\connectivity\netsocket\source\CellularNonIPSocket.cpp:20:
[DEBUG] Output: ./mbed-os/connectivity/netsocket/include/netsocket/CellularNonIPSocket.h:28:10: fatal error: CellularContext.h: No such file or directory
[DEBUG] Output:    28 | #include "CellularContext.h"
[DEBUG] Output:       |          ^~~~~~~~~~~~~~~~~~~
[DEBUG] Output: compilation terminated.
[ERROR] In file included from .\mbed-os\connectivity\netsocket\source\CellularNonIPSocket.cpp:20:
./mbed-os/connectivity/netsocket/include/netsocket/CellularNonIPSocket.h:28:10: fatal error: CellularContext.h: No such file or directory
   28 | #include "CellularContext.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.

Failed to build library

Memory map breakdown for built projects (values in Bytes):
| name | target | toolchain | static_ram | total_flash |
|------|--------|-----------|------------|-------------|

Build failures:
  * DISCO_L475VG_IOT01A::GCC_ARM::MBED-BUILD

[mbed] ERROR: "c:\users\naknai01\appdata\local\programs\python\python37\python.exe" returned error.
       Code: 1
       Path: "C:\Users\naknai01\Desktop\example\New folder\mbed-os-example-blinky"
       Command: "c:\users\naknai01\appdata\local\programs\python\python37\python.exe -u C:\Users\naknai01\Desktop\example\New folder\mbed-os-example-blinky\mbed-os\tools\test.py -D MBED_TEST_MODE -t GCC_ARM -m DISCO_L475VG_IOT01A --source . --build .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM --test-spec .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\test_spec.json --build-data .\BUILD\tests\DISCO_L475VG_IOT01A\GCC_ARM\build_data.json -n tests-test-group-simple-test -v --greentea"
---

jeromecoutant, 2020-11-09:
> and apply patch from [ARMmbed/mbed-os#13809](https://github.com/ARMmbed/mbed-os/pull/13809)

?

Nakul93, 2020-11-09:
Ok sorry for this but how do I apply this patch from CLI using git?

closed 2020-12-10 by Nakul93