ARMmbed/mbed-os-tools #247: Thread isAlive renamed to is_alive in Python 3.9

LDong-Arm, 2020-11-12:
### Description

`mbedgt` doesn't work with Python 3.9:
```
mbedgt: running 1 test for platform 'NRF52840_DK' and toolchain 'ARMC6'                                                                                                                                           
        use 1 instance of execution threads for testing                                                                                                                                                           
mbedgt: could not generate test report'Thread' object has no attribute 'isAlive'                                                                                                                                  
mbedgt: completed in 1.53 sec
```

It's because `isAlive` has been renamed to `is_alive`.

https://github.com/ARMmbed/mbed-os-tools/blob/bb7f07fcf8352e89f2197ca0d0891a91c4e4e281/packages/mbed-greentea/mbed_greentea/mbed_greentea_cli.py#L907

Version:
```
$ mbedgt --version
1.7.4
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

tirkarthi, 2020-11-17:
I have created https://github.com/ARMmbed/mbed-os-tools/pull/248 as a fix

LDong-Arm, 2020-11-17:
> I have created #248 as a fix

Thanks!

PR #248 (2021-09-03): Use is_alive in favour of isAlive for Python 3.9 compatibility.
### Description

Use `is_alive` in favour of `isAlive` for Python 3.9 compatibility. `is_alive` is present in both Python 2 and 3. Fixes #247 

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers
closed 2021-09-03 by rwalton-arm