ARMmbed/mbed-os-tools #257: Testing Cellular drivers/libraries

CalvoM, 2021-04-26:
### Description

I am currently writing integration tests for our network components. We are using SIMCOM modem, i.e. SIM5360E, and thus we extended the Cellular Library to support SIMCOM. However, when we write the integration tests for the network functionality, the integration do not go past the __svcXX parts e.g. the __svcMutex found [here](https://github.com/ARMmbed/mbed-os/blob/a3be10c976c36da222517abc0cb4f81e88ff8552/cmsis/CMSIS_5/CMSIS/RTOS2/RTX/Source/rtx_mutex.c#L526). So, we sought to see how the already built Cellular libraries like [Ublox](https://github.com/ARMmbed/mbed-os/tree/a3be10c976c36da222517abc0cb4f81e88ff8552/connectivity/drivers/cellular/UBLOX) are tested, so that we could find guidance, but we found none. Thus my question, does anyone have links for tests for such cellular libraries so that we could find help or even better, how to solve this svc issue we have in testing?  
We are using PCB with NUCLEO-F429ZI chip and GCC_ARM toolchain.
Thanks in advance.

### Issue request type
    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2021-04-26:
This might be better to report in Mbed OS repository where we would ask connectivity team to review.

CalvoM, 2021-04-26:
Okay lemme do that.