ARMmbed/mbed-os-tools #268: Running greentea not working (Windows)

Ripitchip00, 2021-08-02:
### Description

I'm trying to set up Greentea for testing through [this example](https://os.mbed.com/docs/mbed-os/v6.12/debug-test/greentea-for-testing-applications.html).
- Target : Disco-L475VG-IOT01A
- Toolchain : ARM_GCC (from [here ](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads): version 9-2020-q2-update (I took the gcc-arm-none-eabi-9-2020-q2-update-win32.exe to do the installation)
- mbed-os : tag mbed-os-6.10.0 (tried with latest version of master tho)
- mbed-cli : 1.10.5 
- mbedgt : 1.8.10
- Python v. 3.6.5 (no python 2.7)

The commands I did : 
- `mbed new first-greentea-test`
- `mbed config GCC_ARM_PATH "C:\Program Files (x86)\GNU Arm Embedded Toolchain\9 2020-q2-update\bin"`
- `mbed test -t GCC_ARM -m auto -v -n tests-test-group-simple-test`

I get the following error : 
```
[DEBUG] Return: 1
        [DEBUG] Output: arm-none-eabi-g++: fatal error: cannot execute 'c:/program files (x86)/gnu arm embedded toolchain/9 2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../a
rm-none-eabi/bin/as.exe': CreateProcess: No such file or directory
        [DEBUG] Output: compilation terminated.
```

The program as.exe exists in this path though.

I tried to setup GNU ARM in the PATH as environment variable but it didn't work. I tried to have a shorter path, it didn't work.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

Patater, 2021-08-03:
Have you tried placing the toolchain in a path without spaces in it? Sometimes spaces in paths trip up various scripts and programs.

Ripitchip00, 2021-08-03:
Hi, thanks for your answer.

I tried a path without space and a shorter one but I've got the same problem :
```
[DEBUG] Return: 1
        [DEBUG] Output: arm-none-eabi-g++: fatal error: cannot execute 'c:/users/name.lastname/9-2020-q2-update/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/as.exe': Cr
eateProcess: No such file or directory
        [DEBUG] Output: compilation terminated.
```

The problem is when I copy-paste this path in Windows File Explorer, it's opening a cmd window and executing as.exe. So I don't get why I have this error.

Ripitchip00, 2021-08-03:
Welp, I tried on Linux as well to see how it would work.

It works (at least, the tests are running), but not consistently.

First possible error (with `host test result() call skipped, received: ioerr_serial`) : 
```
mbed-host-test-runner: started
[1627989343.09][HTST][INF] host test executor ver. 1.8.8
[1627989343.09][HTST][INF] copy image onto target...
[1627989343.10][COPY][INF] Waiting up to 60 sec for '076402216E3B650D517FF354' mount point (current is '/media/smartfarming/DIS_L4IOT')...
[1627989351.72][HTST][INF] starting host test process...
[1627989351.72][CONN][INF] starting connection process...
[1627989351.72][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1627989351.72][CONN][INF] initializing serial port listener...
[1627989351.72][PLGN][INF] Waiting up to 60 sec for '076402216E3B650D517FF354' serial port (current is '/dev/ttyACM0')...
[1627989351.72][HTST][INF] setting timeout to: 60 sec
[1627989351.74][SERI][INF] serial(port=/dev/ttyACM0, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1627989351.74][SERI][INF] reset device using 'default' plugin...
[1627989351.85][SERI][INF] waiting 1.00 sec after reset
[1627989352.85][SERI][INF] wait for it...
[1627989352.85][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1627989352.85][CONN][INF] sending up to 5 __sync packets (specified with --sync=5)
[1627989352.85][CONN][INF] sending preamble '1645c40a-2e12-401e-b428-f2b85a07a115'
[1627989352.88][SERI][TXD] {{__sync;1645c40a-2e12-401e-b428-f2b85a07a115}}
[1627989352.99][CONN][RXD] bedmbedmbedmbedmbedmbedmbedmbed
[1627989353.04][CONN][INF] found SYNC in stream: {{__sync;1645c40a-2e12-401e-b428-f2b85a07a115}} it is #0 sent, queued...
[1627989353.04][HTST][INF] sync KV found, uuid=1645c40a-2e12-401e-b428-f2b85a07a115, timestamp=1627989353.038188
[1627989353.12][SERI][ERR] read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
[1627989353.12][HTST][ERR] connection lost, serial.read(2304): read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
[1627989353.12][HTST][WRN] stopped to consume events due to __notify_conn_lost event
[1627989353.12][HTST][INF] __exit_event_queue received
[1627989353.12][HTST][INF] test suite run finished after 1.40 sec...
[1627989353.12][HTST][INF] CONN exited with code: 0
[1627989353.12][HTST][INF] No events in queue
[1627989353.12][HTST][INF] host test result() call skipped, received: ioerr_serial
[1627989353.12][HTST][WRN] missing __exit event from DUT
[1627989353.12][HTST][INF] calling blocking teardown()
[1627989353.12][HTST][INF] teardown() finished
[1627989353.12][HTST][INF] {{result;ioerr_serial}}
mbedgt: retry mbedhtrun 1/1
mbedgt: ['mbedhtrun', '-m', 'DISCO_L475VG_IOT01A', '-p', '/dev/ttyACM0:9600', '-f', '"BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM/TESTS/test-group/simple-test/simple-test.bin"', '-d', '/media/smartfarming/DIS_L4IOT', '-c', 'default', '-t', '076402216E3B650D517FF354', '-r', 'default', '-C', '4', '--sync', '5', '-P', '60'] failed after 1 count
mbedgt: checking for GCOV data...
mbedgt: mbed-host-test-runner: stopped and returned 'IOERR_SERIAL'
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: tests-test-group-simple-test
        test case: group-simple-test
mbedgt: test on hardware with target id: 076402216E3B650D517FF354
mbedgt: test suite 'tests-test-group-simple-test' .................................................... IOERR_SERIAL in 10.56 sec
        test case: 'group-simple-test' ............................................................... ERROR in 10.56 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.1366282379
mbedgt: test suite report:
| target                      | platform_name       | test suite                   | result       | elapsed_time (sec) | copy_method |
|-----------------------------|---------------------|------------------------------|--------------|--------------------|-------------|
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | tests-test-group-simple-test | IOERR_SERIAL | 10.56              | default     |
mbedgt: test suite results: 1 IOERR_SERIAL
mbedgt: test case report:
| target                      | platform_name       | test suite                   | test case         | passed | failed | result | elapsed_time (sec) |
|-----------------------------|---------------------|------------------------------|-------------------|--------|--------|--------|--------------------|
| DISCO_L475VG_IOT01A-GCC_ARM | DISCO_L475VG_IOT01A | tests-test-group-simple-test | group-simple-test | 0      | 1      | ERROR  | 10.56              |
mbedgt: test case results: 1 ERROR
mbedgt: completed in 10.86 sec
mbedgt: exited with code 1
[mbed] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "/home/smartfarming/Mbed Programs/first-greentea-test"
       Command: "mbedgt --test-spec ./BUILD/tests/DISCO_L475VG_IOT01A/GCC_ARM/test_spec.json -n tests-test-group-simple-test -V"
---
[... 64 lines cut ...]
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.1301454205
mbedgt: test suite report:
| target                      | platform_name       | test suite                   | result | elapsed_time (sec) | copy_method |
[... cut ...]
mbedgt: completed in 11.13 sec
```

Linux is on a virtual machine but the USB port is directly connected to the computer (no hub). I installed the extension package and setup usb 3.0 in virtual box parameters. I dunno if this problem would come from virtual machine or not but I didn't see any usb-deconnexion from mbed IDE.

- Linux 18.04 LTS
- Virtual box 6.1.26
- mbed-cli : 1.10.5
- mbedgt : 1.8.10
- Python3 v. 3.6.9
- Python v. 2.7.17

Would be nice to have greentea working in at least one OS :)

jeromecoutant, 2021-08-16:
Hi
Solution is not tricky, you need to use  .mbedignore feature
See https://github.com/ARMmbed/mbed-os/issues/13598

Ripitchip00, 2021-08-18:
Hi, 

Thanks @jeromecoutant, but this wasn't the right answer as I already had a .mbedignore file. 

But I finally found what went wrong, even though I don't understand the logic behind. At beginning, I followed the example here : https://os.mbed.com/docs/mbed-os/v6.13/debug-test/greentea-for-testing-applications.html, where it's said to create a new project with `mbed new first-greentea-test`. 
After a while, I realized that if I create a project normally with _new project_ like I usually do, it works perfectly fine...

I'll close here !